nturley/netlistsvg #7: Allow manual placement of pins

mithro, 2017-12-23:
It would be good if you could specify manual location overrides for things in the generated diagram.

I mainly want this to force the input / output pins to be at a specific location.

I need this so that I can do heirachical generation of diagrams. IE Generate a diagram for the internals of a black box, then embedded that as a skin inside the next level up.

It would also be good for when you want to more control on generation.

nturley, 2017-12-24:
Presumably the positioning information comes from the netlist?

```json
"ports": {
    "clk100": {
        "direction": "input",
        "bits": [ 2 ],
        "x":47,
        "y":42
    },
```
I'm a little confused how that helps you with hierarchy but I'll take your word for it.

kasbah, 2018-06-13:
@nturley are you saying it should already respect x and y provided in the netlist? It doesn't seem so but just wanted to clarify. You meant "Presumably the positioning information _would come_ from the netlist?".