nturley/netlistsvg #11: helpful error messages for invalid netlists

nturley, 2017-12-24:
Right now the script just dies in an unhelpful way or it just makes a mess of the diagram. Ideally, it would explain what went wrong.

nturley, 2019-04-04:
- [ ] Add a JSON schema to check if the JSON is well formed
- [ ] Add sanity checks before parsing the netlist (Do port directions and port connections match? Does the port list match the template port list?)
- [ ] Throw exceptions during module flattening when unusual results are discovered

mithro, 2019-04-04:
FYI - We are also interested in a JSON schema for the Yosys output - https://github.com/SymbiFlow/ideas/issues/7

mithro, 2019-04-04:
So does @drom

nturley, 2019-04-04:
I started it. Its pretty straightforward. netlistsvg may have different fields that are required or optional than other applications though so we may not necessarily be able to single source it.

https://github.com/nturley/netlistsvg/blob/hierarchy/lib/yosys.schema.json

nturley, 2019-04-07:
I think [this](https://github.com/nturley/netlistsvg/blob/master/lib/yosys.schema.json) is pretty much complete from what I can tell. It's JSON5 instead of JSON so I could add comments (probably should change the file extension), and the error messages section I believe isn't part of the JSON schema standard. I think it's a nonstandard extension from ajv-errors. I didn't bother with descriptions and examples, but I may add them later.

nturley, 2019-04-10:
I renamed the file extension, so now it is here:
https://github.com/nturley/netlistsvg/blob/master/lib/yosys.schema.json5