nturley/netlistsvg #17: Port swapping

nturley, 2017-12-24:
There are some pins whose position are arbitrary. The layout engine should be able to move them around to reduce wire crossings.

nturley, 2017-12-29:
Right now all nodes have fixed port positions. Instead, you can set the node to have port constraints of fixed sides, and then adjust the port margin and spacing so that they line up with the original port positions. I think this would slightly change the positioning of the generic ports.

dvc94ch, 2018-06-04:
I don't think there is enough information in the input netlistsvg to find the signal path (probably requires a simulator to figure it out) The orientation of components is not something arbitrary that can be rotated at will.

This seems different from what this issue is about port swapping, which I think is useful. But what I think nturley is talking about is using a side constraint which would work very nicely for binary operators.