nturley/netlistsvg #26: Indicate bus width

kasbah, 2018-01-04:
Would be nice to be able to easily see the width of buses other than the splits and constants.

nturley, 2018-01-04:
ELK supports edge labels, but it's going to really clutter up the schematic if we have every single edge labeled. So I think the only tricky part about this is deciding which edges to add labels to.

Yosys also provides net names, so in theory we could also consider putting the netnames as an edge label.

kasbah, 2018-01-04:
I could have sworn there was some universal convention in digital logic schematics to indicate bus widths with some slanted  edges. Something like this:

[image]

But maybe dotted or squiggly somehow? I can't find any examples. Do you know what I am thinking of? This would have to be done with a graph node.

kasbah, 2018-01-04:
I guess the convention is just a single slanted line and a number next to it. 

[image]

from: https://tex.stackexchange.com/questions/125460/drawing-bus-width-like-markers-with-number-on-arrows

nturley, 2019-03-22:
finding the coordinates of the slanty line also makes this a little tricky.

rjordans, 2019-10-22:
Another option may be to add the slanted line with number as one of the nodes on the line for ELK to place for you?  That should help in keeping things clean

nturley, 2019-10-25:
I like that solution. It would add more columns but I think that is fine.

stevenbell, 2020-11-23:
This would be really helpful to me.  Even just using a bold line to indicate a bus versus a single wire would be a big help.

I'm happy to submit a PR if you can point me in the right direction to get started.

nobodywasishere, 2021-03-19:
I got double line thickness to indicate a bus working with [this commit](https://github.com/nobodywasishere/netlistsvg/commit/7dc47a6aa10b30be236f8fe689b5134419925853). 

Before:

[image]

After (along with some other changes I made):

[image]

I've also started [looking into](https://github.com/nobodywasishere/netlistsvg/commit/c2300a7d2f6ba6d0bf8f2ef41e1acdf82013c5a0) indicating the bus width with an [edge label](https://www.eclipse.org/elk/reference/options/org-eclipse-elk-edgeLabels-inline.html) but so far it's proved unsuccessful. ElkJS seems to make space for the labels but doesn't actually place them for some reason.

nturley-copia, 2021-03-19:
That looks great! Well done

nturley, 2021-03-19:
Looks great. Well done.

rjordans, 2021-03-19:
that's looking nice already!

nobodywasishere, 2021-03-19:
I'm also on the [netlistsvg Gitter](https://gitter.im/netlistsvg/Lobby) if anyone wants to talk about this there.

mithro, 2021-03-19:
This is super cool!

mithro, 2021-03-19:
I would love to see this integrated into netlistsvg and demos in https://github.com/SymbiFlow/sphinxcontrib-hdl-diagrams

PR #101 (2021-03-25): Bus width labels
Add bus labels and bus width. Fixes #26. Buses have double line thickness and a label indicating their width.

Each line also now has a class for its width (`width_*`) and each label has a class for the width it's representing (`busLabel_*`). This allows more fine tuned configuration in a skin file or after the fact in an outputted SVG. Examples added to README.md.

## Example 1

Before:
[image]

After:
[image]

After with example modifications from README:
[image]

## Example 2

Before:
[image]

After:
[image]
closed 2021-03-25 by nturley