nturley/netlistsvg #33: pass orientation in netlist and auto-rotate skins

dvc94ch, 2018-01-10:
This would lead to a deduplication of schematic symbols

kasbah, 2018-05-31:
Any ideas on how to achieve the auto-rotation?

nturley, 2018-06-02:
Its easy to rotate the symbol but you'll also need to transform the port coordinates. I think thats the only tricky part. But I havent spent much time thinking about it.

kasbah, 2018-06-03:
I misunderstood what `auto-rotate` means here. I  am interested in auto orientation of components based on the connections. The only way I can think of doing this at the moment would be to iterate through several layouts while rotating symbols and finding some sort of optimum. (number of layers minimization? edge crossing minimization?)

dvc94ch, 2018-06-06:
Another tricky part is getting the text right. Currently the skins don't have refs and values, but you don't want them rotated in the same way as the symbol.