nturley/netlistsvg #35: pass arbitrary attributes to skin

dvc94ch, 2018-01-10:
Setting classes on symbols would be great. This means that only one diode or transistor symbol is required.
The diode skin would look like this:
```svg
<g s:type="diode" s:width="20" s:height="50" transform="translate(75,280)">
   <s:alias val="d"/>
   <path d="M0,15 H20 L10,35 Z M0,35 H20" class="symbol"/>
   <path d="M10,0 V15 M10,35 V50" class="connect"/>
   <!-- schottky -->
   <path s:attr="schottky" d="M0,35 V40 M20,35 V30" class="symbol"/>
   <g s:x="10" s:y="0" s:pid="+"/>
   <g s:x="10" s:y="50" s:pid="-"/>
</g>
```

nturley, 2019-03-22:
Doesn't this already work? Is this not needed now that #38 has gone in?

nturley, 2019-04-20:
I think this is done now. So I'm going to close it for now.

closed 2019-04-20 by nturley