nturley/netlistsvg #36: annotate ports traces and junctions with a net id

dvc94ch, 2018-01-14:
This would simplify mapping the schematic parts to the pcb.

amstan, 2019-03-30:
Yes please! I want this a lot for my uses.

A way for a label to show up on a net.
I don't think there's even a way to provide net name right now in the schema, just node number.

amstan, 2019-03-30:
Once that's done. It would be nice to show nets that only connect to one place but is labeled/named.

This way you could do airwires:
[image]

nturley, 2019-04-01:
This is actually very doable because yosys netlists contain net names and ELK supports edge labels. So should be pretty straightforward.

amstan, 2019-04-13:
Cool. I saw [hints in the schema](https://github.com/nturley/netlistsvg/blob/07e5c18950729dc9e297cf2de157ba02cd7866e1/lib/yosys.schema.json5#L77) about this.

So I guess we'll be able to do:

	"netnames": {
		"PP3300": { "bits": [ 1 ] }
		"GND": { "bits": [ 2 ] }
		"VBUS": { "bits": [ 3 ] }
		....
	}

The name bits is not very semantic for analog, but I assume it's something dictated by yosys.

How would the output of elkjs be processed to keep the net names nearby? I guess a <text> element would need to be emitted, plopping it down somewhere. Ideally without routing other wires through it.

PS: Why does the schema also talk about "0" and "1" instead of the node numbers? What does that allow us to do?

nturley, 2019-04-20:
> How would the output of elkjs be processed to keep the net names nearby?

ElkModel.wireNameLookup associate ELK edge IDs to netNames (ie. "6", "24,25", etc). See https://github.com/nturley/netlistsvg/blob/master/lib/elkGraph.ts#L195

> I guess a element would need to be emitted, plopping it down somewhere

ELK supports edge labels, so that part is pretty straightforward. See https://www.eclipse.org/elk/documentation/tooldevelopers/graphdatastructure/jsonformat.html

> Why does the schema also talk about "0" and "1" instead of the node numbers?

Digital yosys constants (http://www.clifford.at/yosys/cmd_write_json.html) 

> Signal bits that are connected to a constant driver are denoted as string "0" or "1" instead of
a number.

Apparently now Yosys supports also supports "x".

Perhaps the skin could indicate which nodes to generate for special string values? Maybe "0" should indicate ground on the analog skin for example.

amstan, 2020-03-10:
I had a go at this: https://github.com/amstan/netlistsvg/commit/d4d9742fe131b8bee083f9faa49ec114ad862e96

Now I need to look at airwires, but I think it might be hard since it's essentially an edge with only one node connected to it, going nowhere.

nturley, 2020-03-11:
There are a few cases where we create invisible nodes for wire routing purposes. ELK supports nets connected to one or more input ports and one or more output ports but I needed nets connected to zero or more input or output ports. So when there is zero input or zero output ports I create a tiny dummy node and erase it after layout and I erase the wire all the way to the nearest junction. Kind of a nasty hack but it produced really good results.

amstan, 2020-03-11:
I tought more about it, I don't know if I even want elk to do the layout for this.

The thing is, airwires, especially when just coming out of one big chip, always look like some kind of comb. Straight out, parallel with eachother and to the same distance.

I'm thinking the effect could be simulated by drawing it manually after the fact at elk to svg generation time. The extra padding around the chip could be achieved by telling elk the chip's port labels are really long.

Otherwise what do you think of my [change](https://github.com/amstan/netlistsvg/commit/d4d9742fe131b8bee083f9faa49ec114ad862e96) so far? It's one of my first serious attempts with javascript/typescript.

amstan, 2020-03-12:
> I'm thinking the effect could be simulated by drawing it manually after the fact at elk to svg generation time. The extra padding around the chip could be achieved by telling elk the chip's port labels are really long.

And here we are!
http://hypertriangle.com/~alex/upload/test.svg

[image]

Code: https://github.com/amstan/netlistsvg/commit/d49c1d3ca4705ee53f48b7f6d71514e10f109d57

nturley, 2020-03-15:
That image looks pretty good. Well done.

nturley, 2020-03-15:
And your change looks pretty good as well too. Honestly, even reading my own typescript at this point is making me cringe. This was my first experience with typescript and since then I've used it for several work projects and so I've learned a whole lot since I wrote this.