nturley/netlistsvg #37: Hierarchal designs

kasbah, 2018-01-15:
This stood out to me on the README before but was removed in some re-shuffling and I couldn't see an issue for it. 

>Yosysjs and klayjs are both capable of handling sub circuits and subgraphs. It might be interesting to see what kind of images I can produce from a hierarchal design.

I guess the more interesting thing to me would be to allow navigating hierarchies in an interactive fashion but that may be out of scope for this project.

dvc94ch, 2018-01-15:
> This stood out to me on the README before but was removed in some re-shuffling and I couldn't see an issue for it.

I think this would probably be better as a separate project. My current thinking on this is that we feed each subckt into netlistsvg and then click the ports to open the right netlist inside the viewer.

> I guess the more interesting thing to me would be to allow navigating hierarchies in an interactive fashion but that may be out of scope for this project.

I'm working on `pcbsvg` currently which is preparation for such a viewer. Even if placement and routing is done in kicad, `pcbsvg` should be useful to check whether electrogrammar matched a CPL part. 

This looks like a starting point: https://github.com/nturley/nturley.github.io

I think that the editor component should be removed and something like browser sync used instead, so that any editor can be used.

kasbah, 2018-01-20:
Interactive viewer will be very cool. Netlistsvg should be able to handle hierarchical netlists from Yosys. So far I haven't been able to generate any though. Can anyone tell me how to use `yosys -p "prep"` on hierarchal designs. Running it on the [DE2i-150 example](https://github.com/YosysHQ/yosys/tree/master/examples/intel/DE2i-150) yields:

```
yosys> prep;

6. Executing PREP pass.

6.1. Executing HIERARCHY pass (managing design hierarchy).
ERROR: Module `\sevenseg' referenced in module `\top' in cell `\UUD0' is not part of the design.
```

And I am getting a similar thing for any hierarchal designs.

kasbah, 2018-01-20:
Ok, I guess I don't need to run "prep". Running `yosys -p "hierarchy; write_json" top_ver.v -o top.json` on [this example](https://cloud.monostable.co.uk/hierarchal.zip) (taken from [here](https://www.altera.com/support/support-resources/design-examples/design-software/verilog/ver_hier.html)) yields following JSON. I guess it doesn't actually read the bottom* Verilog files. 😕

```json
{
  "creator": "Yosys 0.7 (git sha1 61f6811, gcc 5.4.0-6ubuntu1~16.04.4 -O2 -fstack-protector-strong -fPIC -Os)",
  "modules": {
    "top_ver": {
      "attributes": {
        "src": "top_ver.v:1"
      },
      "ports": {
        "q": {
          "direction": "input",
          "bits": [ 2 ]
        },
        "p": {
          "direction": "input",
          "bits": [ 3 ]
        },
        "r": {
          "direction": "input",
          "bits": [ 4 ]
        },
        "out": {
          "direction": "output",
          "bits": [ 5 ]
        }
      },
      "cells": {
        "u2": {
          "hide_name": 0,
          "type": "bottom2",
          "parameters": {
          },
          "attributes": {
            "src": "top_ver.v:8"
          },
          "connections": {
            "n": [ 5 ],
            "m": [ 4 ],
            "l": [ 6 ]
          }
        },
        "u1": {
          "hide_name": 0,
          "type": "bottom1",
          "parameters": {
          },
          "attributes": {
            "src": "top_ver.v:7"
          },
          "connections": {
            "c": [ 6 ],
            "b": [ 3 ],
            "a": [ 2 ]
          }
        }
      },
      "netnames": {
        "intsig": {
          "hide_name": 0,
          "bits": [ 6 ],
          "attributes": {
            "src": "top_ver.v:5"
          }
        },
        "out": {
          "hide_name": 0,
          "bits": [ 5 ],
          "attributes": {
            "src": "top_ver.v:4"
          }
        },
        "r": {
          "hide_name": 0,
          "bits": [ 4 ],
          "attributes": {
            "src": "top_ver.v:3"
          }
        },
        "p": {
          "hide_name": 0,
          "bits": [ 3 ],
          "attributes": {
            "src": "top_ver.v:3"
          }
        },
        "q": {
          "hide_name": 0,
          "bits": [ 2 ],
          "attributes": {
            "src": "top_ver.v:3"
          }
        }
      }
    }
  }
}
```

nturley, 2019-04-18:
#64 is more specific about how to accomplish this so I'm going to close this issue as a duplicate.

closed 2019-04-18 by nturley