nturley/netlistsvg #43: typescript

dvc94ch, 2018-06-08:
Hi

Anyone against using typescript in netlistsvg? Alternatively we could add a index.d.ts just declaring the types. @nturley @kasbah

kasbah, 2018-06-08:
I was actually just thinking about opening an RFC issue for this: I would like to set up a test-suite with jest with [snapshot tests](https://facebook.github.io/jest/docs/en/snapshot-testing.html) and then refactor everything using typescript and properly define all the intermediate object types (because it's pretty confusing the way it is currently). Thoughts on that?

In the meantime it wouldn't hurt to at least declare the types of the API if you can't wait for a refactor.

dvc94ch, 2018-06-08:
no that sounds good. I started hacking on a couple of features a while back and thought it could use a refactoring. that was before I started experimenting with xtext sprotty. Let me know how I can help...

kasbah, 2018-06-08:
First step would be to add a jest test framework with snapshots, if you want to start with that. That's independent of if we switch to TS or not (since @nturley needs to agree as well for that to happen).

dvc94ch, 2018-06-08:
Also I'd like to keep the netlist json around during all steps of the processing. Currently the json is changed into some intermediate representation at the beginning dropping a lot of useful information like source lines.

kasbah, 2018-06-08:
Ok, that's a completely separate issue though. :grin:

nturley, 2018-06-13:
No problem with typescript or jest at all. I've been pretty busy in my personal life right now, so I apologize if I'm not super responsive right now. Don't let me hold you back to try neat things.

For kicks and giggles, I actually started writing it in Rust at one point. It wasn't terrible.

Part of me kinda wants to rewrite the analog netlist layout as a separate project. If we didn't have to worry about this weird compatibility with yosys JSON netlists, I think the first few transformations on the input netlist would be unnecessary. Also, all of the code that handles splits and joins as well as the way we handle constants doesn't make as much sense in analog land. Or how we have to arbitrarily label ports as inputs and outputs. I think if we started from scratch and abandon this compatibility with digital netlists, it would open up a lot of possibilities. Is that a bad idea? I feel like most of the investment in this code has been in knowledge and strategies instead of actual lines of code. It's still not a very large code base.

kasbah, 2018-06-13:
Rust to web assembly or how? I don't really think it makes a lot of sense for something like this (but can see that it would be fun). 

I think freeing the analog part from the yosys netlist constraint does make a lot of sense but I am not sure splitting the analog and digital parts into separate projects is needed. In the end they do share a pretty similar goals even with their differences. A few features like #37 and maybe #17 will benefit both. 

I think we _should_ come up with a good intermediate representation of our graphs/netlists and then transform yosys (and other netlists) to that format.

dvc94ch, 2018-06-13:
Mmh I think we should keep the netlist format. I'm working on an analog
extension and a mixed signal netlist/rtl language. I'm using an extended
yosys netlist as a target for the frontend, and plan on using a transform
in the backend to split mixed modules into digital/analog modules use yosys
to convert the digital part to verilog or spice and merge with the spice
generated from the analog portion.

kasbah, 2018-06-13:
I don't think anyone was suggesting removing support for Yosys netlists.

kasbah, 2018-06-13:
On a second read, I guess nturley kind of was, but just for analog portions if it seems to be getting in the way.

dvc94ch, 2018-06-13:
What I'm saying is that we need to support a well  defined superset of
yosys netlists and stop making a distinction with different skins and
options. I don't see a reason why flip-flops and resistors can't be in the
same netlist or resulting schematic.

kasbah, 2018-06-13:
Maybe we should move this to a chat or something because it has very little to do with moving to typescript but if it's a _super_ set anyway then why not design a better format more suited to our needs (and transform yosys netlists to that format when needed)?

We have seen vastly very different requirements for mixed signal schematics to purely digital ones so I can't see why or how we should stop making a distinction. No reason why you can't have a flip-flop in the "analog" skin though.

kasbah, 2018-06-13:
Continued discussion in case anyone is interested: https://gitter.im/monostable/electro-grammar?at=5b219bd18864a936cc863750

nturley, 2018-08-12:
I started a PR to migrate to typescript. Take a look at it and see what you think. I'm still working on the refactoring.

nturley, 2019-03-20:
I merged the typescript PR. It should have no functional changes besides the addition of the dev dependencies. There's still a lot of work to clean it up, but most of the function arguments are now typed.

I've never really done any typescript before so I'm still figuring things out, but so far it's looking like it's a little easier to follow than the JS version was.

closed 2019-03-20 by nturley

nturley, 2019-03-27:
jest tests are starting to go in now as well.