nturley/netlistsvg #45: Add a unique id to the generated elements

Badel2, 2018-10-04:
Hi, it would be really useful to be able to use `getElementById` to modify the SVG after it has been generated. My use case: I am building a web-based digital logic simulator, and I am using netlistsvg to render the components and interact with them. My guess is that many people may want to use javascript to make the graphs interactive.

I managed to figure out how to add onClick listeners to the inputs, and change the color of the wires on the fly. Showcase: [SRLatch GIF](https://imgur.com/RWHww0e), [web demo](https://badel2.github.io/comphdl/demo/v09/?example=example1.txt&top=Mux_4_1).

You can see the commits on [my fork](https://github.com/Badel2/netlistsvg), they are very short as I'm only adding a unique id:

Inputs/outputs: https://github.com/nturley/netlistsvg/commit/683b882fb005e13bdc1667f6734e53b22cd8748f
Wires: https://github.com/nturley/netlistsvg/commit/1c0ecd891fda9a61b8d3bfeb0557af962513d456

With wires it's a bit more complicated because they are made out of many segments, so I decided to add the same `class` tag and use CSS to change the color.

But basically, I'm only missing a way to be able to select components by id, simply getting the name of the component when I click on it would be enough for my purposes.

Thanks!

nturley, 2019-03-25:
I added an attribute of the node id to generated nodes, I'll be adding them to edges soon. Some of the hyper edge cases confused me. Also, our edge ID names are not super useful names right now, so I'm hoping to add the netnames from the netlist as well if they are available.

amstan, 2019-03-30:
It would be awesome to have css classes(ex: `.net-VCC`) set for each net (either via net name, or node number). This way one could a whole net with the same color via CSS.

Same for parts (`#part-U3`) and each pin of a part (`#pin-U3.VCC`).

With a simple `:target {color: #ffff99;}` css line, now one could link to a specific part in the document: `my_circuit.svg#part-U3` and have that part get highlighted (and ideally automatically scrolled to, given a nice browser).

Another use case for this is to embed the svg in a bigger html document. Then one could add javascript to make all parts/nets clickable, cross-linkable.

nturley, 2019-04-10:
think I got it. The rules of the game are

* all cells have an id in the format of "cell_" + cell id
* generic ports have an id in the format of "port_" + port id + "~" + cell id
* any elements in a cell template in the skin can include "$cell_id" in the class and that will get replaced with the id of the cell.
* nets have a class in the format of "net_" and then a comma separated list of signal numbers.

for example (with name spaced properties and nodes removed for clarity)
```
<!-- generic -->
<g transform="translate(519,87.5)" id="cell_businterface">
    <text x="15" y="-4" class="nodelabel cell_businterface">businterface</text>
    <rect width="30" height="40" class="cell_businterface"/>
    <g transform="translate(0,10)" id="port_businterface~clk">
      <text x="-3" y="-4" class="inputPortLabel cell_businterface">clk</text>
    </g>
    <g transform="translate(0,30)" id="port_businterface~reset">
      <text x="-3" y="-4" class="inputPortLabel cell_businterface">reset</text>
    </g>
  </g>
<!-- external port -->
<g transform="translate(531,283.5)" id="cell_clk125">
    <text x="15" y="-4" class="nodelabel cell_clk125">clk125</text>
    <path d="M30,0 L30,20 L15,20 L0,10 L15,0 Z" class="cell_clk125"/>
    <g />
  </g>
<!-- wire -->
  <line x1="40" x2="70" y1="217.5" y2="217.5" class="net_2"/>
```

There's still some more work I could do on this, but I think this is enough to call the issue closed for now. Feel free to open more issues for further work in this area.

closed 2019-04-10 by nturley

amstan, 2019-04-10:
This is great!
https://google.github.io/pcbdl/examples/servo_micro.withsvg.html#cell_U59(SN74AVC4T774RSVR)

I need more javascript for simulating target on something like a net. Probably more javascript will be able to make things clickable. But that's more of my problem :)

I also thought I would have to remove my "refdes(value)" cell naming, but looks like my browser is happy enough going to weird things like `#cell_R502(4.7kΩ)`. Otherwise, I don't really know how to make netlistsvg label the passives with anything other than the cell name.