nturley/netlistsvg #46: very large diagrams get limited to 8000px wide

mattvenn, 2019-01-04:
I'm trying to get a layout of the picorv32 riscV processor drawn.
It works! (with increased node stack size!).
However, the output image is very long and thin. 8000x31000px.

Is there a way to specify an aspect ratio? I took a look at the elks documentation and couldn't find anything obvious. I grepped the source code for 8000 and only found memory limits.

Thanks for the great resource!

mattvenn, 2019-01-05:
hmm, tried a few more and the 8000 was just a one off. Can't change the issue name for some reason.

nturley, 2019-03-20:
This is a very late reply. But better late than never, I suppose.

I'd experiment with the layering strategy first to see if network simplex or something makes it a bit more compact. IIRC network simplex more evenly distributes nodes, but I think it has the same number of layers (horizontal columns) as longest path.

Next I'd look at some of the other ELK variables to force the horizontal width to be smaller. horizontal space between nodes, maximum number of ranks, and so on.

If none of the ELK parameters work out well enough, then I'd try to modify the actual elements in the skin file to be more narrow. It would be pretty easy to apply a geometric transformation on each of the nodes, but you'll also need to modify the metadata that specifies the port locations.