nturley/netlistsvg #49: ports should also support multiple aliases

nturley, 2019-03-26:
allowing multiple aliases for the same symbol allows us to not duplicate the same symbols for every name that yosys might give it. Having different cell types with different port names means you can't reuse the same symbol.

nobodywasishere, 2021-03-19:
Is this implemented? I can see in the default.svg several aliases listed for each symbol.
```svg
<g s:type="mux" transform="translate(50, 50)" s:width="20" s:height="40">
  <s:alias val="$pmux"/>
  <s:alias val="$mux"/>
  <s:alias val="$_MUX_"/>
```

mithro, 2021-03-19:
@nobodywasishere -- I think that is aliases for the object type rather than the ports?

nturley-copia, 2021-03-19:
Correct. I was running into cases where the symbol should've been able to be reused but the ports had different names.