nturley/netlistsvg #50: Internal loader.js error

rhthomas, 2019-03-26:
Running netlistsvg on my machine fails with the below error message, however when the generated JSON file is loaded into the online demo, it renders the netlist fine.

```
> yosys -p "prep -top top; write_json dance.json" top.v div.v mem.v pc.v
...
> node ../tools/netlistsvg/bin/netlistsvg.js dance.json -o dance.svg

internal/modules/cjs/loader.js:651
    throw err;
    ^

Error: Cannot find module '../built'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:649:15)
    at Function.Module._load (internal/modules/cjs/loader.js:575:25)
    at Module.require (internal/modules/cjs/loader.js:705:19)
    at require (internal/modules/cjs/helpers.js:14:16)
...
```

I've noticed this was changed in commit `decf30`. I have tried uninstall and reinstall of npm, as well as running the makefile in the examples folder. All result in the same error.

nturley, 2019-03-27:
node can't run this by itself anymore it needs to be built by running tsc. I'll either need to update the instructions, or include the build in the repo. The longterm solution is to get the built js files from npm but checking the build into the repo should be a good stopgap until then.

nturley, 2019-03-27:
The built folder is included in the repo so it should work again now. Let me know if you have any more problems.

closed 2019-03-27 by nturley

rhthomas, 2019-03-27:
All working perfectly. Cheers!