nturley/netlistsvg #51: Error: start and end arent orthogonal

amstan, 2019-03-30:
I'm using your wonderful project to draw smaller sections of my schematic.

I keep getting "Error: start and end arent orthogonal" depending on how I slice my circuit or how many things I decide to put in the circuit.

	amstan@amstan-desktop:~/examples (master*)% nodejs ~/netlistsvg/bin/netlistsvg.js --skin ~/netlistsvg/lib/analog.svg out_bad.json
	Error: start and end arent orthogonal
	at which_dir (/home/amstan/netlistsvg/built/drawModule.js:68:15)
	at /home/amstan/netlistsvg/built/drawModule.js:132:24
	at Array.map (<anonymous>)
	at _loop_1 (/home/amstan/netlistsvg/built/drawModule.js:120:30)
	at removeDummyEdges (/home/amstan/netlistsvg/built/drawModule.js:156:23)
	at Object.drawModule [as default] (/home/amstan/netlistsvg/built/drawModule.js:18:5)
	at /home/amstan/netlistsvg/built/index.js:25:57

[examples.zip](https://github.com/nturley/netlistsvg/files/3025016/examples.zip)

See out_bad.json for an example that would fail it.
I eventually tracked it down to the last 2 parts (GND430 and C5). If I remove them (out_good.json is exactly that), then netlistsvg is happy with it.
Another workaround was to make the cap's (C5) port_directions both output.

nturley, 2019-04-01:
The error message means that it was trying to trace a wire it got from the ELK layout and for orthogonal routing all lines should be vertical or horizontal but for some reason it was diagonal. I'll keep looking at it. Possibly related to #47?

nturley, 2019-04-04:
It looks like this is an unusual case where a dummy node edge had multiple junctions and the bends and the junctions weren't in the same order. The bends still appear to be in the order from startPoint to endPoint, but the junctions are not.
```
{
      "id": "e31",
      "source": "$d_0",
      "sourcePort": "$d_0.p",
      "target": "PP3300389",
      "targetPort": "PP3300389.A",
      "sections": [
        {
          "id": "e31_s0",
          "startPoint": {
            "x": 92,
            "y": 47
          },
          "endPoint": {
            "x": 82,
            "y": 42
          },
          "bendPoints": [
            {
              "x": 92,
              "y": 57
            },
            {
              "x": 82,
              "y": 57
            }
          ]
        }
      ],
      "junctionPoints": [
        {
          "x": 82,
          "y": 57
        },
        {
          "x": 92,
          "y": 57
        }
      ]
    },
```

amstan, 2019-04-04:
Oh! That sounds like something I noticed before!

If you open [my svg](https://google.github.io/pcbdl/examples/servo_micro.svg) in something like chrome, zoom out all the way, you'll notice some lines are darker than others. That's because there's multiple lines going the same route sometimes, it's as if they don't know they started from the same spot.

Ex: `<line x1="1744" x2="1744" y1="1534.5" y2="1717"/>` is on top of another line, try changing its x1 coord and you'll see another one just behind it.

nturley, 2019-04-05:
[image]

I think I fixed it but there was also something confusing in your netlist that threw it off. The undocumented rule for analog netlists (until I fix the port directions bug) is that all downward facing ports must be outputs and all upward facing ports must be inputs. horizontal port direction doesn't matter.

So this port should be an output.
```
"PP3300389": {
  "connections": {"A": [389]},
  "port_directions": {"A": "input"},
  "type": "vcc"
},
```
By connecting VCC to two other input ports, it caused it to create a dummy node as an output to make it a valid hyper edge which is why you ended up with an extra junction and weird routing.

Now that I fixed the bug, it shouldn't crash but it still would do some odd things if you break the rule.

closed 2019-04-05 by nturley

amstan, 2019-04-05:
Yay! Thank you. I can't get it to ever error anymore.

Once I fixed my directions, things look to be rendering properly too.

Weirdly, I had to remove my swap pin port direction logic now too: https://github.com/google/pcbdl/commit/9ed5e8bf4fc9e3a5d26aeb28c8454231a2b85f38#diff-526fb11c69a6225770197dff49f9efeaL204

I guess it only likes "A" being on top and "B" on the bottom now.

I also noticed if I temporarily disable pin swapping, it gets really angry with me and shows artifacts again. I originally added that pin swapping code to prevent the kind of port swapping issues also described in https://github.com/dvc94ch/pycircuit/issues/14#issuecomment-356339584. It is possible now that all those examples from pycircuit without port swapping now have artifacts too.

nturley, 2019-04-05:
I just fixed #47 so now all port directions are ignored except for generic node ports which need them to know whether to put them on the left or right. Eventually generic node ports won't be set to fixed positions and then generic nodes won't need them either.

amstan, 2019-04-07:
One more artifact that I found:

A vertical resistor between 2 grounds.
Or... A vertical resistor to ground, a vertical capacitor also to ground, the tops of them connected together with a wire.

Both of those cases yield on the part on the left (either the second gnd or the capacitor/resistor) not really connected to a wire, just a little farther from it, and a junction where the horizontal wire starts.

I guess it's because we're connecting 2 top facing (therefore inputs) together.

nturley, 2019-04-07:
Ah, I messed it up. I filed a new issue #59

nturley, 2019-04-10:
okay, the rewrite of removeDummyEdges is done now with two simple test cases (basically subsets of hyper edges.json). I'm sure there's more funny edge cases I haven't thought of yet, so if you find any more then we can add those to the test cases.