nturley/netlistsvg #52: Redundant "Larry" VCCs and other connections

amstan, 2019-03-30:
I keep seeing connections that are redundant. Note a lot of the VCCs, they connect one way, but then they go around the part and connect from the other side too, just to be sure. Skin doesn't seem to make a difference for my file.

[image]
See attachment for source json:
[out_good.zip](https://github.com/nturley/netlistsvg/files/3025022/out_good.zip)

I noticed the same thing happening when using the [common emitter example](https://raw.githubusercontent.com/nturley/netlistsvg/master/examples/analog/common_emitter_full.json) in the [demo](https://nturley.github.io/netlistsvg/) with the analog skin selected.

I remember there used to be a rendered picture of the common_emitter circuit in the docs and it didn't have this problem.

nturley, 2019-04-01:
This might be an issue I've seen before. If you look at the first screenshot on the readme you'll see there's a redundant line to count_out. It's due to the way that ELK doesn't really handle hyper edges.

I probably made it even worse when I added "lateral port" handling a while back.

I'll need to add a step to search for these redundant wires and delete it down to the minimum number required to connect all of the ports.

kasbah, 2019-04-01:
Are you setting postCompaction now? https://github.com/nturley/netlistsvg/pull/40#issuecomment-394457005

nturley, 2019-04-04:
I need to write up a document of all of the types of bad layouts.

I think ELK is tracking this issue here: https://github.com/eclipse/elk/issues/374

I just added postCompaction the other day but it probably won't help with this particular issue.

amstan, 2019-04-04:
Here's the latest output (pulled a few hours ago) for my schematic: https://google.github.io/pcbdl/examples/servo_micro.svg

You can still see the same problem around U7 and U3.

postCompaction did make a lot of things shorter, I like that.

nturley, 2019-04-05:
It looks like youd benefit from wide nodes on multiple layers. We should try turning that on the analog skin as well. https://www.eclipse.org/elk/reference/options/org-eclipse-elk-layered-layering-widenodesonmultiplelayers.html

amstan, 2019-04-05:
I feel like I'm doing something wrong when trying to enable this.

It has no effect. But to be fair, even if I had typos it wouldn't complain about it

	diff --git a/lib/analog.svg b/lib/analog.svg
	index 0ef2d2e..6956616 100644
	--- a/lib/analog.svg
	+++ b/lib/analog.svg
	@@ -9,6 +9,7 @@
		org.eclipse.elk.layered.spacing.nodeNodeBetweenLayers="5"
		layered.compaction.postCompaction.strategy="4"
		elk.spacing.nodeNode= "35"
	+        org.eclipse.elk.layered.layering.wideNodesOnMultipleLayers="AGGRESSIVE"
		org.eclipse.elk.direction="DOWN"/>
	<s:low_priority_alias val="$dff" />
	</s:properties>

nturley, 2019-04-05:
odd, I would've thought that would do something more interesting from the description.

amstan, 2019-04-05:
I can only assume I didn't enable it properly.