nturley/netlistsvg #53: Add text/values to parts (especially skinned)

amstan, 2019-03-30:
I would like to label my resistors, capacitors with reference designators (R1, C1) and values (1k, 100nF).

Right now there's no way to add any text except by changing the type to something custom (to disable the skin), even then, would be nice to support multiple lines.

For the VCC part in the analog skin, I would like to change the text "VCC" dynamically, for designs that have more than one voltage rail. Perhaps that could be the VCC part's "value".

Another smaller request that might be related: for blocky chip type parts, a way to put the pin names inside and pin numbers outside. For now my workaround is to just have the pin numbers in () and live with the fact that everything's outside of the block.

amstan, 2019-03-30:
I'm really sorry, this looks to be handled by https://github.com/nturley/netlistsvg/pull/38

amstan, 2019-03-30:
Nevermind.

I saw this picture and got excited, but we don't do that yet:
[image]

We would need to edit the default skin file to have those attributes for the resistors and other parts.

nturley, 2019-04-01:
Yeah, ideally each cell template should be able to add as many text boxes as it likes and indicate which attributes from the netlist cell to pull them from. I think that's where we were going with #35?

nturley, 2019-04-03:
ah, I see what happened here. It was completed but never added to the analog skin. I'll add them back.

nturley, 2019-04-04:
In addition to just adding the labels to the skin, there's some additional work to getting the labels into ELK so that layout routes around them.

amstan, 2019-04-04:
Looks good so far! Thanks!

closed 2019-04-19 by nturley