nturley/netlistsvg #58: How to generate from an NGSPICE netlist?

nturley, 2019-04-07:
sent via email
--------
It's possible to use the "analog.svg" library to draw a analog schematic from NGSPICE netlist?

In other words, how to convert a NGSPICE netlist to yosys JSON netlist ?

If you think that this is possible, may you give me an example of a yosys JSON netlist equivalent to this simple NGSPICE netlist (using "analog.svg") ? (don't need to be complete, just a partial functional example can help me to understand the rest)

Vcc 1 0 dc 5
Vs 50 0 0.0 ac 1 sin(0 2 10k)
Rs 50 2 100
RL 3 0 2k
Q1 4 2 0 QNPN
R1 1 4 2.0e3
C1 4 3 2.0e-9

[image]

nturley, 2019-04-07:
Without making any changes to the skin, this is the best I could do to reproduce your example.
[image]

```
{
  "modules": {
    "Analog Example": {
      "ports": {},
      "cells": {
        "Vs": {
          "type":"v",
          "connections": {
            "+": [5],
            "-": [0]
          }
        },
        "Vcc": {
          "type":"v",
          "connections": {
            "+": [1],
            "-": [6]
          }
        },
        "C1": {
          "type":"c_h",
          "connections": {
            "A": [4],
            "B": [3]
          }
        },
        "Rs": {
          "type": "r_h",
          "connections": {
            "A": [5],
            "B": [2]
          }
        },
        "R1": {
          "type": "r_v",
          "connections": {
            "A": [1],
            "B": [4]
          }
        },
        "RL": {
          "type": "r_v",
           "connections": {
             "A": [3],
             "B": [7]
           }
        },
        "Q1": {
          "type": "q_npn",
          "connections": {
            "C": [4],
            "B": [2],
            "E": [8]
          }
        },
        "gnd0": {
          "type": "gnd",
          "connections": {
            "A": [0]
          }
        },
        "gnd6": {
          "type": "gnd",
          "connections": {
            "A": [6]
          }
        },
        "gnd7": {
          "type": "gnd",
          "connections": {
            "A": [7]
          }
        },
        "gnd8": {
          "type": "gnd",
          "connections": {
            "A": [8]
          }
        }
      }
    }
  }
}
```

nturley, 2019-04-07:
sent via email
--------
Now it's clear for me:
Each cell is a symbol in the schematic;

* The cell's type bind to the skin `"s:type"`; 
* Cell's port_directions bind each component pin to the skin `"s:pid"`
* Cell's connections bind each component pin to a Spice net label.

closed 2019-04-07 by nturley

qpwo, 2024-05-10:
Could you share your script for converting spice netlist to json?