nturley/netlistsvg #63: Weird position of DLUT in diagram

mithro, 2019-04-17:
The json at https://gist.github.com/mithro/97fed031fe3fbbf5e2bf687bebf4e275 produces the following diagram;
[image]

I'm trying to figure out why it decided to place the `DLUT` at the bottom of the image near the `CARRY0_CONST` object? It Seems like it would have been a much nicer diagram is all the D objects where placed above the `CLUT`.

Any idea on how to improve this diagram?

nturley, 2019-04-17:
There's a few problems, but this isn't the worst I've seen. But to your specific question, I don't think I can do much about the vertical positioning of `DLUT`. That's deep in the ELK layout engine and I can't find any knobs to twiddle in regards to vertical positioning sweeps (https://www.eclipse.org/elk/reference/options.html). I'm hoping to clean up the duplicate path around `AFFMUX` but that would be after ELK has run, so I can only remove elements, not reposition them. So I think this is as good as we can do except hope the ELK developers can find ways to improve their algorithm or replace ELK with our own layout engine. I have been experimenting with some alternative techniques to classical layered graph layout and it's possible someday I may have something that will do a better job (but probably much slower).

I've read some of the ELK papers but I'm not an expert in their code. I can make some guesses about why it made this particular choice from what I know about how layered graph layout algorithms usually work, (which I'll be discussing at my presentation at Latch Up). This particular node seems like it's a pretty pathological case for the vertical node ordering step.

Vertical ordering is usually determined by the median vertical ordering of it's neighbors in the adjacent columns.. It sweeps left to right and right to left a few times and moves the nodes toward the relative positions of it's neighbors on the right or to the left depending on which direction it is sweeping. The only interesting neighbor of `DLUT` on it's left side is `CARRY0_CONST`, which seems to be dragging it downward, so for left to right sweeps that's probably how it got down there in the first place. Right to left sweeps should've had a chance at pulling back up though.

At this point in the layout algorith, I believe `DLUT` would consider itself to have 3 neighbors to the right-side. Two of which are real nodes (`F7BMUX`, `DCY0`), the third is a hidden node (or "dummy node") that is an intermediate destination to route the edge to the columns further to the right. It should be a little under `DCY0`. The median of those three points is at `DCY0`, so it looks like a right to left sweep wouldn't do much good either.

So it seems to be doing exactly what the classical strategy for layered graph layout is for vertical node ordering. This looks like a pretty fundamental problem for layered graph layout algorithms. They do a really good job at minimizing horizontal edge distance, but not as good of a job for vertical edge distance.

mithro, 2019-04-17:
What confuses me is that the `DLUT` isn't connected to the `CARRY0_CONST` in any way?

nturley, 2019-04-17:
Ah sorry, I thought those were connected. So the only left neighbors are external ports. Odd. Regardless I still dont know what we can do about it. Unless I'm building my ELK JSON incorrectly or there is an ELK layout parameter that helps then we have to accept the layout it gives us.

mithro, 2019-04-17:
I'm afraid I don't know much about how ELK works, so don't know if any of these are a possibility;

 * Tell ELK to align the LUTs in a vertical line in the D, C, B, A order?
 * Increase the force which pushes unrelated items away from each other?
 * Increase the negative weighting of having lines which cross each other?

mithro, 2019-04-17:
@nturley I have a feeling that this diagram use to be much better laid out by older versions of netlistsvg.. I might try running some tests.

mithro, 2019-04-17:
There are actually a lot of things that "feel weird" about the layout of this diagram. I have marked a bunch on the diagram below;
[image]

Any idea why the inputs for the LUTs are not being pulled next to the LUTs? It seems to be affecting the `CLUT` and `DLUT` while the `BLUT` and `ALUT` look good.

mithro, 2019-04-17:
Is there a "maximum height" or "maximum aspect ratio" configured for ELK that could be preventing the layout engine from doing a better job?

nturley, 2019-04-18:
You could try putting this in the skin https://www.eclipse.org/elk/reference/options/org-eclipse-elk-aspectratio.html

but I doubt it will help. It seems like the problem is the vertical ordering of the nodes, not the maximum total height.

mithro, 2019-04-18:
Is there a way I can force an ordering around key parts?

mithro, 2019-04-18:
@nturley I just discovered http://nturley.github.io/ which lets me drag the components around and such. How is the dragging interacting with ELKJS? 

Maybe I could feed it the above JSON file and drag the items around to see if I can tease the diagram into better shape?

nobodywasishere, 2021-07-03:
How does this diagram compare to the original one you posted? This is with the latest version of ELKjs, which has a few layout improvements. `C6` is the only really weird one I can see.

[image]