nturley/netlistsvg #64: Support for nested netlists?

mithro, 2019-04-17:
Looking at the ELK page they have this diagram;
[image]

Looking at the channel 1 and channel 2 sections, this diagram shows support for "diagram inside a diagram" type thing.

It would be really cool to be able to nest a netlist inside another netlist and use this feature. Any idea how to support this and how hard it would be to add?

nturley, 2019-04-18:
I was tracking that here: #37 
You can see some of the work I did on it here: https://github.com/nturley/netlistsvg/commit/fc1d6e847ff98350e94d651219d5f2203d8c69af

It's not too bad complexity-wise, but it's going to require a little bit of refactoring because the code just wasn't organized very well for it.

nturley, 2019-04-18:
This is a little more specific than #37, so I think I'll close #37 as a dupe and leave this one open.

mithro, 2019-04-18:
@nturley Oh cool! Looking forward to it!

qarlosalberto, 2020-12-20:
What is the progress of this feature? I would like to add it to my project. Its really awesone! :)