nturley/netlistsvg #74: Recent versions output invalid SVGs

amstan, 2019-05-29:
I bisected this and ever since https://github.com/nturley/netlistsvg/commit/a953939b77e201c187c2f52821a09d253eaacdbd landed, the SVG files are a little corrupt. they have things like [object Object] in them (near the <style> tag).

<details><summary>Here's a diff of the output</summary>

``` diff
	diff --git a/examples/servo_micro.i2c.svg b/examples/servo_micro.i2c.svg
	index 24f69f0..b4325dc 100644
	--- a/examples/servo_micro.i2c.svg
	+++ b/examples/servo_micro.i2c.svg
	@@ -1,5 +1,6 @@
	<svg xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" xmlns:s="https://github.com/nturley/netlistsvg" width="567.02" height="207">
	-  <style>svg {
	+  <style,[object Object],
	+  svg {
		stroke: #000;
		fill: none;
	}
	@@ -28,7 +29,8 @@
		stroke-linejoin: round;
		stroke-linecap: round;
		fill: #000;
	-  }</style>
	+  }
	+  >
	<g s:type="generic" s:width="30" s:height="40" transform="translate(12,152)" id="cell_U6(STM32F072CBU6TR)">
		<text x="15" y="-4" class="nodelabel cell_U6(STM32F072CBU6TR)" s:attribute="ref">U6 (STM32F072CBU6TR)</text>
		<rect width="30" height="40" x="0" y="0" s:generic="body" class="cell_U6(STM32F072CBU6TR)"/>
	@@ -114,4 +116,36 @@
	<line x1="62" x2="358" y1="132" y2="132" class="net_139"/>
	<line x1="358" x2="358" y1="132" y2="152.5" class="net_139"/>
	<line x1="358" x2="441" y1="152.5" y2="152.5" class="net_139"/>
	+  </style,[object Object],
	+  svg {
	+    stroke: #000;
	+    fill: none;
	+  }
	+  text {
	+    fill: #000;
	+    stroke: none;
	+    font-size: 10px;
	+    font-weight: bold;
	+    font-family: "Courier New", monospace;
	+  }
	+  .nodelabel {
	+    text-anchor: middle;
	+  }
	+  .inputPortLabel {
	+    text-anchor: end;
	+  }
	+  .splitjoinBody {
	+    fill: #000;
	+  }
	+  .symbol {
	+    stroke-linejoin: round;
	+    stroke-linecap: round;
	+    stroke-width: 2;
	+  }
	+  .detail {
	+    stroke-linejoin: round;
	+    stroke-linecap: round;
	+    fill: #000;
	+  }
	+  >
	</svg>
```
</details>

nturley, 2019-05-30:
Yeah, I can see what happened. CI has inkscape parsing the SVG, so I would've expected this to fail CI, but inkscape must be a pretty forgiving SVG parser. After I fix this, I'll add a unit test to validate the SVG output so that things like this don't happen again.

nturley, 2019-08-08:
@amstan can you confirm this is fixed now with #77 ?

amstan, 2019-08-08:
Yes. Seems to work for me for stuff that used to fail.

Thanks @AubreyEAnderson

closed 2019-08-08 by amstan