nturley/netlistsvg #78: Add at least one git tag?

mithro, 2019-10-13:
Could you add a git tag to the repository? If you don't have versions it would be good to add `v0.0` to the first ever commit.

This way `git describe` will report useful output when used on the netlistsvg repository something like `v0.0-888-g23345`.

mithro, 2019-10-13:
Looks like you might want;
 * `git tag -a v1.0.0 714698506f4d3b8f6942c5433880b22afeed4fed` and 
 * `git tag  -a v1.0.1 63013e2aad0ea493a751c97484b443549f3e2ea0`

eine, 2020-11-11:
At the same time, it would be desirable to release a new version to npm. Currently, the website can handle multiple netlists which are broken with the version installed through npm.

nturley, 2020-12-12:
boom! Only took me like a year.

closed 2020-12-12 by nturley