nturley/netlistsvg #82: INOUT Port direction throws error

bit0fun, 2019-11-06:
This is a feature request to add support for INOUT ports as currently an error message saying "should be equal to one of the allowed values" is thrown in relation to the direction defintion as the allowedValues contain only "input" and "output". The generation of the json file was from yosys, but I'm not sure if there is anything I have incorrectly generated. The error in question is provided in the image below. 

[image]

I have little understanding of javascript, otherwise I would have attempted to create a patch for this feature.

If there is any more information that is needed, feel free to ask.

nturley, 2019-11-07:
By way of explanation, this particular error is being raised by the schema validator. To avoid weird errors further down the processing chain, we check to see if the json we are provided matches our schema to make sure that the file is formatted in a way that we expect.

Our schema says that we only support port directions of input and output, which is why it correctly errored.

I think we can add support for inouts without too much effort. I think the right thing to do is assign an inout port some direction, maybe to consider all inout ports to be output ports. For the purposes of layout that would put inout ports on the right-hand side of generic nodes which feels about right.

bit0fun, 2019-11-07:
Ok so that assumption was more or less correct; yes the error of course should have been thrown.

I would agree that the right side would make sense, for flow reasons. However, for something like a large bus, maybe having the connections on the top or bottom may make more sense? I would think that most inout ports are buses or multinode communication lines, therefore the diagram would possibly be a bit cleaner by only putting the connections in a drop configuration. Maybe this could be conditional to whether or not more than two ports are connected to the same net?

SmallRoomLabs, 2019-12-07:
I +1 this request since I have some modules with INOUT's that are giving me grief when trying to graph them...

rw1nkler, 2020-09-21:
I'm also interested in this topic!

nobodywasishere, 2021-03-19:
I've found that adding `inout` to the schema makes netlistsvg treat it like an output (which is to be expected due to [these lines](https://github.com/nturley/netlistsvg/blob/eb9dc546beae573d98635a7b9d9c5d12af3f695b/built/Cell.js#L31-L35)), and I've not run into any issues with it, though I've not tested it too extensively. What potential issues are there with this that I'm missing?

nturley-copia, 2021-03-19:
Should be fine. There's a lot of logic in there to handle a wire with multiple outputs but no inputs so assuming all of that is still working then that should be the only change necessary.

nturley, 2021-03-19:
I think that's it. There's already logic in there to handle multiple outputs and no inputs for a wire so assuming all of that still works then we should be fine.

PR #102 (2021-03-28): Multimux, tribuffer, better icons, and inout support
- Modifies most current icons to make them more even and work better with thicker line strokes
- Adds a multimux if bus width input is greater than 1
- Line endings are now rounded
- Adds tribuffer icon
- Fixes #82: adds `"inout"` and `"z"` to `lib/yosys.schema.json5`, which are treated like `"output"` and `"x"`, respectively

## Example 1

Before:
[image]

After:
[image]

## Example 2

Before:
[image]

After:
[image]
closed 2021-03-28 by nturley