nturley/netlistsvg #85: Wires overlap for simple drawing.

joseluisquiroga, 2020-02-28:
The wires for first and last ports overlap as if they were a single wire for the json code generated.

The json code was generated with the following command:
yosys -p "read_verilog recu.v; prep -top unary_nand -flatten; aigmap; write_json recu.json"

For the file recu.v containging the following code:

module unary_nand
#(parameter WIDTH = 32)
(input [WIDTH-1:0] in_and,
output            out_and);

generate
 if(WIDTH == 1) begin
   assign out_and = in_and;
 end
 else if(WIDTH == 2) begin
   assign out_and = in_and[0] & in_and[1];
 end
 else begin
   unary_nand #(.WIDTH (WIDTH/2))
     unary_nand_low
       (.in_and  (in_and[WIDTH/2-1:0]),
        .out_and (out_and_low));

   unary_nand #(.WIDTH (WIDTH - WIDTH/2))
     unary_nand_high
       (.in_and  (in_and[WIDTH-1:WIDTH/2]),
        .out_and (out_and_high));

   assign out_and = out_and_low & out_and_high;
 end
endgenerate
endmodule

The json code is:

{
  "creator": "Yosys 0.9+932 (git sha1 613334d9, gcc 7.4.0-1ubuntu1~18.04.1 -fPIC -Os)",
  "modules": {
    "unary_nand": {
      "attributes": {
        "dynports": 1,
        "top": 1,
        "src": "recu.v:4"
      },
      "ports": {
        "in_and": {
          "direction": "input",
          "bits": [ 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33 ]
        },
        "out_and": {
          "direction": "output",
          "bits": [ 34 ]
        }
      },
      "cells": {
        "$auto$aigmap.cc:123:execute$10": {
          "hide_name": 1,
          "type": "$_AND_",
          "parameters": {
          },
          "attributes": {
          },
          "port_directions": {
            "A": "input",
            "B": "input",
            "Y": "output"
          },
          "connections": {
[... 1356 lines cut ...]
        "unary_nand_low.unary_nand_low.unary_nand_low.unary_nand_high.out_and": {
          "hide_name": 0,
          "bits": [ 64 ],
          "attributes": {
            "src": "recu.v:17|recu.v:22|recu.v:7"
          }
        },
        "unary_nand_low.unary_nand_low.unary_nand_low.unary_nand_low.in_and": {
          "hide_name": 0,
          "bits": [ 2, 3 ],
          "attributes": {
            "src": "recu.v:17|recu.v:6"
          }
        },
        "unary_nand_low.unary_nand_low.unary_nand_low.unary_nand_low.out_and": {
          "hide_name": 0,
          "bits": [ 63 ],
          "attributes": {
            "src": "recu.v:17|recu.v:7"
          }
        }
      }
    }
  }
}

amstan, 2020-04-13:
Could you put your code in a:
\<details>\<summary>whatever.json\</summary>long code here\</details>

nobodywasishere, 2021-07-03:
I think this is closer to what you were wanting (even if I can't see what you originally had), though there are still a few overlapping wires around the edges of the split block. This is with a newer version of ELKjs.
[image]

nturley, 2021-07-05:
duplicate #105

closed 2021-07-05 by nturley