nturley/netlistsvg #90: What's the meaning of "connections"?

haolimin118, 2020-04-16:


kasbah, 2020-04-16:
This may be documented properly somewhere but simply put, you list the nets each "port" (i.e. pin) connects to. A simple analog example:

[image]

Is specified like this:

```json5
{
    modules: {
        example: {
            ports: {
                vcc: {
                    direction: "input",
                    bits: [
                        1
                    ]
                }
            },
            cells: {
                gnd: {
                    type: "gnd",
                    port_directions: {
                        A: "input"
                    },
                    connections: {
                        A: [
                            2
                        ]
                    }
                },
                r1: {
                    type: "r_v",
                    port_directions: {
                        A: "input",
                        B: "output"
                    },
                    connections: {
                        A: [
                            1
                        ],
                        B: [
                            2
                        ]
                    }
                }
            }
        }
    }
}
```

kasbah, 2020-04-16:
The confusing thing, and I can't remember why it's like this (haven't used netlistsvg in quite a while) is in an analog schematic you can only add one net, i.e. one number into `A` in e.g. `connections: {A: [1]}`.

closed 2021-07-04 by nturley