nturley/netlistsvg #91: Error: should be equal to one of the allowed values

adumont, 2020-05-08:
Hi,
In local on my computer, I have updated netlistsvg (npm install -g netlistsvg), when I run this command I get:

```
netlistsvg netlist.json -o assets/vgaLabels.svg
/usr/local/lib/node_modules/netlistsvg/bin/netlistsvg.js:50
            throw Error(JSON.stringify(ajv.errors, null, 2));
            ^

Error: [
  {
    "keyword": "enum",
    "dataPath": "/modules/vgaLabels/attributes/top",
    "schemaPath": "#/properties/modules/additionalProperties/properties/attributes/properties/top/enum",
    "params": {
      "allowedValues": [
        0,
        1
      ]
    },
    "message": "should be equal to one of the allowed values"
  }
]
    at parseFiles (/usr/local/lib/node_modules/netlistsvg/bin/netlistsvg.js:50:19)
    at fs.readFile (/usr/local/lib/node_modules/netlistsvg/bin/netlistsvg.js:37:13)
    at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:504:3)

```

In case that can help, here's the [netlist.json](https://termbin.com/h3rx)

When tested on the demo site, it doesn't fail (although it doesn't seem to render very correctly either).

I have tested with different modules (.v files), different designs, I always get the same error.

Another module [here](https://termbin.com/z6rp), also failing now.

What could I try? (not an expert in node.js/npm stuff :( )

amstan, 2020-05-09:
Looks like the the yosys schema needs to be updated perhaps (or the thing you're outputting is not conforming to the schema (i'm not sure, i don't know the tooling)). Affected spot on netlistsvg's end: https://github.com/nturley/netlistsvg/blob/master/lib/yosys.schema.json5#L94

From the looks of it, if you get rid of this block it should work:
```
      "attributes": {
        "dynports": "00000000000000000000000000000001",
        "top": "00000000000000000000000000000001",
        "src": "XALU.v:6.1-54.10"
      },
```

I'm sorry, I couldn't actually reproduce the problem and try the fix quickly, the [online demo](https://nturley.github.io/netlistsvg/) seems to render your input file without problems. I guess it might be an older version that wasn't as strict about the schema as the last version of netlistsvg.

amstan, 2020-05-09:
Here's a bigger bug that explains more about what's going on: https://github.com/nturley/netlistsvg/issues/87

djwubs, 2020-05-09:
This is indeed a duplicate of issue #87. This will be fixed if #88 is merged. For now, you can change the value for top manually, so  
```json
      "attributes": {
        "dynports": "00000000000000000000000000000001",
        "top": "00000000000000000000000000000001",
        "src": "XALU.v:6.1-54.10"
      },
```
should then become
```json
      "attributes": {
        "dynports": "00000000000000000000000000000001",
        "top": 1,
        "src": "XALU.v:6.1-54.10"
      },
```
If you are using Yosys to generate the json, then it is also possible to use the `-compat-int` option of the `write_json` command. This will also fix your problem.

adumont, 2020-05-09:
Thanks, that's exactly what I was looking for :-D. Working perfectly now. Usinsg the -compat-int option of the write_json command fixed my problem.

closed 2020-05-09 by adumont