nturley/netlistsvg #93: Unconnected and overlapping wires in a net

spededd, 2020-05-26:
[image]

**Input json file:**
`{"creator":"Yosys 0.5+220 (git sha1 94fbaff, emcc  -Os)","modules":{"gates":{"cells":{"and_2_1_1590516389917":{"type":"$and","hide_name":0,"port_directions":{"A":"input","B":"input","Y":"output"},"connections":{"A":[3],"B":[1],"Y":[4]}},"and_2_2_1590516404134":{"type":"$and","hide_name":0,"port_directions":{"A":"input","B":"input","Y":"output"},"connections":{"A":[5],"B":[3],"Y":[6]}},"and_2_3_1590516426292":{"type":"$and","hide_name":0,"port_directions":{"A":"input","B":"input","Y":"output"},"connections":{"A":[7],"B":[6],"Y":[8]}},"and_2_4_1590516432644":{"type":"$and","hide_name":0,"port_directions":{"A":"input","B":"input","Y":"output"},"connections":{"A":[9],"B":[5],"Y":[10]}},"PORT9":{"hide_name":0,"type":"$rec","parameters":{"WIDTH":1},"port_directions":{"Y":"output"},"connections":{"Y":[9]}},"and_2_5_1590516445731":{"type":"$and","hide_name":0,"port_directions":{"A":"input","B":"input","Y":"output"},"connections":{"A":[11],"B":[7],"Y":[12]}},"PORT11":{"hide_name":0,"type":"$rec","parameters":{"WIDTH":1},"port_directions":{"Y":"output"},"connections":{"Y":[11]}},"and_2_6_1590516453833":{"type":"$and","hide_name":0,"port_directions":{"A":"input","B":"input","Y":"output"},"connections":{"A":[13],"B":[10],"Y":[14]}},"and_2_7_1590516461836":{"type":"$and","hide_name":0,"port_directions":{"A":"input","B":"input","Y":"output"},"connections":{"A":[15],"B":[8],"Y":[16]}},"PORT15":{"hide_name":0,"type":"$rec","parameters":{"WIDTH":1},"port_directions":{"Y":"output"},"connections":{"Y":[15]}},"PORT16":{"hide_name":0,"type":"$rec","parameters":{"WIDTH":1},"port_directions":{"A":"input"},"connections":{"A":[16]}},"and_2_8_1590516473667":{"type":"$and","hide_name":0,"port_directions":{"A":"input","B":"input","Y":"output"},"connections":{"A":[17],"B":[13],"Y":[18]}},"PORT17":{"hide_name":0,"type":"$rec","parameters":{"WIDTH":1},"port_directions":{"Y":"output"},"connections":{"Y":[17]}},"PORT18":{"hide_name":0,"type":"$rec","parameters":{"WIDTH":1},"port_directions":{"A":"input"},"connections":{"A":[18]}},"and_2_9_1590516482679":{"type":"$and","hide_name":0,"port_directions":{"A":"input","B":"input","Y":"output"},"connections":{"A":[19],"B":[4],"Y":[20]}},"PORT20":{"hide_name":0,"type":"$rec","parameters":{"WIDTH":1},"port_directions":{"A":"input"},"connections":{"A":[20]}},"and_2_10_1590516492145":{"type":"$and","hide_name":0,"port_directions":{"A":"input","B":"input","Y":"output"},"connections":{"A":[21],"B":[19],"Y":[22]}},"PORT21":{"hide_name":0,"type":"$rec","parameters":{"WIDTH":1},"port_directions":{"Y":"output"},"connections":{"Y":[21]}},"PORT22":{"hide_name":0,"type":"$rec","parameters":{"WIDTH":1},"port_directions":{"A":"input"},"connections":{"A":[22]}},"and_2_11_1590516499508":{"type":"$and","hide_name":0,"port_directions":{"A":"input","B":"input","Y":"output"},"connections":{"A":[23],"B":[14],"Y":[24]}},"PORT23":{"hide_name":0,"type":"$rec","parameters":{"WIDTH":1},"port_directions":{"Y":"output"},"connections":{"Y":[23]}},"PORT24":{"hide_name":0,"type":"$rec","parameters":{"WIDTH":1},"port_directions":{"A":"input"},"connections":{"A":[24]}},"and_2_12_1590516507992":{"type":"$and","hide_name":0,"port_directions":{"A":"input","B":"input","Y":"output"},"connections":{"A":[25],"B":[12],"Y":[26]}},"PORT25":{"hide_name":0,"type":"$rec","parameters":{"WIDTH":1},"port_directions":{"Y":"output"},"connections":{"Y":[25]}},"PORT26":{"hide_name":0,"type":"$rec","parameters":{"WIDTH":1},"port_directions":{"A":"input"},"connections":{"A":[26]}}}}}}`

**skin file** - /lib/default.svg

**some debug info:**

1. The issue is caused by net_13, below is the svg snippet

`<line x1="133" x2="133" y1="52.5" y2="85" class="net_13"/>
<circle cx="133" cy="85" r="2" style="fill:#000" class="net_13"/>
<line x1="133" x2="160.66666666666666" y1="85" y2="85" class="net_13"/>
<line x1="133" x2="133" y1="52.5" y2="183" class="net_13"/>
<line x1="133" x2="231" y1="183" y2="183" class="net_13"/> `

2. added offset & line stroke to better view the net. red lines are causing the issue

`<line x1="123" x2="123" y1="52.5" y2="85" class="net_13" stroke="red"/>
  <circle cx="133" cy="85" r="2" style="fill:#000" class="net_13"/>
  <line x1="133" x2="160.66666666666666" y1="85" y2="85" class="net_13"/>
  <line x1="128" x2="128" y1="52.5" y2="183" class="net_13" stroke="red"/>
  <line x1="133" x2="231" y1="183" y2="183" class="net_13"/> `

[image]

spededd, 2020-06-11:
If possible I would like to get some attention for this issue. Thank you.

djwubs, 2020-06-11:
The net is only connected to input ports, which makes it a special case. Currently, nets connected to either only input ports or only output ports are not supported by ELK. As a workaround for this, an invisible dummy node is used to add an output or input port respectively so ELK can generate the layout. After the layout is created, this dummy node and the net leading to the dummy node should deleted. 

What you are seeing here for net 13 is part of the wire leading to the dummy node. As far as I know, this problem is caused by the function `findBendNearDummy()` in drawModule.ts, which should determine the point until which the net should be deleted. I do not fully understand the process of choosing this point, but it is based on the `bendPoints` of the net.

I have experimented a little bit with the `findBendNearDummy()` function to find a solution to this bug. When using the `junctionPoints` of the net, instead of the `bendPoints`, the right schematic is generated, as shown in the image. Despite the fact that it generates the right schematic, I am not sure this is the right solution, as I am not sure about the logic behind it. I also do not know if this is a universal solution for all cases. For reference, a closed issue related to this issue is issue #59 (and maybe issue #23).

[image]