nturley/netlistsvg #94: Error involving "enum" keyword

TheProgrammerIncarnate, 2020-06-06:
When I tried running Netlistsvg on a JSON netlist file, it gave me this error

`
[cole@cole-laptop serv]$ netlistsvg serv_built.json
/usr/lib/node_modules/netlistsvg/bin/netlistsvg.js:50
            throw Error(JSON.stringify(ajv.errors, null, 2));
            ^

Error: [
  {
    "keyword": "enum",
    "dataPath": "/modules/serv_top/attributes/top",
    "schemaPath": "#/properties/modules/additionalProperties/properties/attributes/properties/top/enum",
    "params": {
      "allowedValues": [
        0,
        1
      ]
    },
    "message": "should be equal to one of the allowed values"
  }
]
    at /usr/lib/node_modules/netlistsvg/bin/netlistsvg.js:50:19
    at /usr/lib/node_modules/netlistsvg/bin/netlistsvg.js:37:13
    at FSReqCallback.readFileAfterClose [as oncomplete] (internal/fs/read_file_context.js:63:3)

`
This same error happens with all the other netlists I tested, which are all direct verilog conversions through read>hierarchy>prep>write_json. Interestingly, the web demo version of Netlistsvg works fine for all of them. Also, I didn't find any accourances of the string "enum", "dataPath", "schemaPath", "params", or "allowedValues" in the JSON files

Note: my Netlistsvg installation is up-to-date.

djwubs, 2020-06-06:
This particular bug has already been fixed in commit 67115bcc9147ca32f3135d0a52ccc50980b6dce6. 

I have looked into why it is still possible to run into problems, and it seems that the npm package has not been updated to match this repository. It would be the best to increase the version number and publish it on npm to update the npm package in my opinion, but I am not very experienced with npm. @nturley, can you look into this?

djwubs, 2020-06-06:
For now, you can use one of the two options mention here to bypass this errror: https://github.com/nturley/netlistsvg/issues/91#issuecomment-626126246

TheProgrammerIncarnate, 2020-06-06:
@djwubs Thanks for your quick response! That bypass worked for me.

closed 2020-06-06 by TheProgrammerIncarnate