nturley/netlistsvg #97: Names with dots not rendered.

Zottel, 2020-10-12:
When I render circuits with input/output names containing dots, i.e. from SystemVerilog interfaces, these names are not rendered. Is this a bug or intended behaviour? Is there a way to escape these names or should I remove them from my output?
Looking forward to your reply and thank you for the awesome tool :)

[something.withdots.json.txt](https://github.com/nturley/netlistsvg/files/5364213/something.withdots.json.txt)

[image]

[something.withoutdots.json.txt](https://github.com/nturley/netlistsvg/files/5364214/something.withoutdots.json.txt)

[image]

nturley, 2020-10-16:
netlistsvg uses periods internally for some internal naming so I'm guessing that must've confused it. I didn't intend not to support names with dots though so I suppose it's a bug.

Zottel, 2020-11-27:
Fixed by 2c1fb82.

closed 2020-11-27 by Zottel