nturley/netlistsvg #98: Unable to generate diagram

mickey4u, 2020-10-23:
After obtaining the json file, I tried to generate the svg but follow error messages

```
/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:107
    return (haystack.indexOf(needle) > -1);
                     ^

RangeError: Maximum call stack size exceeded
    at String.indexOf (<anonymous>)
    at arrayContains (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:107:22)
    at /usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:113:16
    at baseFindIndex (/usr/local/lib/node_modules/netlistsvg/node_modules/lodash/lodash.js:802:11)
    at Function.findIndex (/usr/local/lib/node_modules/netlistsvg/node_modules/lodash/lodash.js:7280:14)
    at indexOfContains (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:112:14)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:164:17)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
    at gather (/usr/local/lib/node_modules/netlistsvg/built/FlatModule.js:188:5)
```

nturley, 2020-12-12:
do you have a reproducible example?

Zottel, 2021-08-02:
I have a module that causes the same error.
It's untested work in progress, so I suspect there are still some directionality conflicts or similar that trigger this.

[cache_readonly_dm.txt](https://github.com/nturley/netlistsvg/files/6917392/cache_readonly_dm.txt)

nobodywasishere, 2021-08-13:
Your netlist is too big. I've run into this before when my netlist includes very large memory arrays, for instance. Not sure how to fix this.

Zottel, 2021-08-18:
The `gather` function appears to be mostly tail recursive, so rewriting it into a loop without recursion may be the solution.

kshaa, 2022-04-01:
_Kind of_ managed to refactor `FlatModule.gather` into a loop as opposed to a recursion as per @Zottel 's suggestion in PR #121 , it means that the restriction in that computation is now heap memory as opposed to stack memory. It didn't fix my personal use case as I have a super huge array I guess, but it might help someone elses use case maybe. :shrug:

j2kun, 2022-10-31:
+1 I also ran into this issue. The arrays in my example are not that big, maybe 100-200 entries.