nturley/netlistsvg #100: netlistsvg running on PC does not create circuit svg using json file created for FPGA synthesis

shahzadmuzaffar, 2021-02-14:
Problem: I synthesize a Verilog code for an FPGA and generate a JSON file using yosys. When I use netlistsvg running on a PC to generate an svg circuit file using the JSON file, it throws errors. If I use the same JSON over the website, it generates the circuit successfully.

**Environment used:**
- Tried on both Windows 10 and Manjaro Linux
- latest netlistsvg over github
- latest yosys
- synthesized for Xilinx FPGA with default options

**How to regenerate:**
JSON file is at the following link. Use this file to generate SVG using netlistsvg over a PC. I used the following command.
>> netlistsvg simple_SGLC_xilinx.json -o simple_SGLC_xilinx_Svg.svg --skin "C:\Users\Shahzad Muzaffar\AppData\Roaming\npm\node_modules\netlistsvg\lib\analog.svg"

**Files: Errors file and the JSON file are at:**
https://drive.google.com/drive/folders/1v5ec8U83k8iBMM5grhbu-yOie1eQ-NiX?usp=sharing

nturley, 2021-02-14:
Looks like there's a bidirectional port (inout) which netlistsvg doesn't support. I'm not sure how to draw a bidirectional port. Outputs go to the right and inputs go to the left so I'm not sure which side to put it on.

I'm guessing that synthesis generated an IOBUF or something. Easiest workaround is find all "inouts" in the json and replace them with "output".

Also it looks like you used the analog skin for a digital schematic. You'll probably get better results with the digital skin.

nturley, 2021-02-14:
Duplicate with #82

shahzadmuzaffar, 2021-02-14:
Ok. I understand and will look into this. Thank you for your reply.
But I have another question. If the inout is not supported, then why on the demo page we successfully generate circuit  diagram? The behavior should not be the same for both PC and web? Just curious.

rjordans, 2021-02-19:
As far as I've seen the examples only show in or out individually, the bi-directional inout port type is the one that's not supported.  Things going in a single direction should work fine.

nobodywasishere, 2021-07-03:
Inouts have been added to netlistsvg, and your json compiles without errors. However, there still may be another problem here because there's a lot more in the JSON than what comes out of netlistsvg. Is this what you were looking to generate?
[image]

shahzadmuzaffar, 2021-07-04:
Yes, it is actually. I would like to try the updated version of netlistsvg on my PC. Should I update it using "npm update -g netlistsvg" command?

Thank you for the update and your kind support.

nobodywasishere, 2021-07-04:
You'll have to install from source (for now) as there hasn't been a release with the updated changes yet. 

To do that on Windows, you should be able to run this (having installed nodejs, npm, and git). Let me know if this doesn't work (as I've only tested it on Linux).
```sh
npm uninstall -g netlistsvg # remove old version
git clone https://github.com/nturley/netlistsvg
cd netlistsvg
npm install # install dependencies
npm install -g . # install netlistsvg to system
```

rjordans, 2021-07-04:
You can usually also install packages from github directly with npm:

`npn install nturley/netlistsvg`

closed 2021-07-05 by nturley