nturley/netlistsvg #104: Add skins for missing Yosys RTL cells

nobodywasishere, 2021-04-02:
Some cells from Yosys that could use a skin. I created this issue to figure out which cells should get skinned, as well as what skin they should get. I'm pulling information from the latest version of the [Yosys manual](https://github.com/YosysHQ/yosys/blob/master/manual/CHAPTER_CellLib.tex). All ideas, comments, and constructive criticism welcome.

### unary operators
- [ ] `$pos`, making into a plus sign may get confusing with addition
- [ ] `$neg`, making into a minus may get confusing with subtraction
- [ ] `$reduce_and` should be added as an alias to `and`
- [ ] `$reduce_or` and `reduce_bool` should be added as aliases to `or`

### binary operators
- [ ] `$shl`, `$shr`, `$sshl`, `$sshr` could be done with just `<<`,  `>>`, `<<<`, and `>>>` respectively
- [ ] `$mul`,  `*` or `x` could work
- [ ] `$div`, `/` could work
- [ ] `$mod`, `%` could work
- [ ] `$pow`, `^` or `a^b` may work
- [ ] `$eqx` may be added as an alias to `$eq`
- [ ] `$ne` and `$nex` could have a `=/=` symbol

### registers
- [ ] `$sr`, I don't think this one needs a skin
- [ ] `$adff`, same as `$dff` but have an `ARST` (asynchronous reset) port. This could simply just be on the bottom of the current `$dff` skin
- [ ] `$sdff`, same as `$dff` but have an `SRST` (synchronous reset) port. This could simply just be on the bottom of the current `$dff` skin, though not sure how to distinguish from `$adff`, maybe by putting an `A` or `S` next to the reset port
- [ ] `$dlatch`, `$dlatchsr`, and `$adlatch`, not sure what this one would look like
- [ ] `$dffsr`, not sure what this one would look like

There are several other register variants listed in the manual that may be too niche / difficult to properly define a skin for. These are `$dffe`, `$adffe`, `$dffsre`, `$sdffe`, and `$sdffce`.

### memories
- [ ] `$mem`, `$memrd`, `$memwr`, `$meminit` - I'm not sure what to do for these, if anything should be done

mithro, 2021-04-02:
@nobodywasishere - I'm sure I had some of these already created somewhere but I can't seem to find them now....

mithro, 2021-04-02:
Apparently I had some stuff at https://github.com/mithro/symbiflow-arch-defs/commit/bd58242a08c838492a7b30533aeece218a64c37a?short_path=e0d91d8#diff-e0d91d811dd290812325d4ed53ad63fc429631785ea6edaf4c835205aef59563

mithro, 2021-04-02:
[image]

mithro, 2021-04-02:
There seems to be some documentation at https://github.com/mithro/symbiflow-arch-defs/blob/4ac22bdc2f6261c09db5f00034b52c66d5f482ce/vpr/latches/README.md

mithro, 2021-04-02:
https://raw.githubusercontent.com/mithro/symbiflow-arch-defs/f47a4e1539c844aac164692f4fdb164f19740861/common/svg/default.svg

[image]

mithro, 2021-04-02:
https://raw.githubusercontent.com/mithro/symbiflow-arch-defs/277b176d04243067af4bd37f12be03347275593f/common/netlistsvg.svg

mithro, 2021-04-02:
And on the topic of flip flops and latches - I have a giant table of flip flops and latches found in various FPGA architectures at https://docs.google.com/spreadsheets/d/16yvScRkedOkPCRBSInFuHE_cUMD8Ls-b9KFmHocpC-o/edit#gid=0 and some details about the mapping between them at https://docs.google.com/spreadsheets/d/1pWjU8RujiGdlr-8iG66O1rvognNHgtnNVvCjSfksH-4/edit#gid=0

mithro, 2021-04-02:
There is also some code to generate flip flops from the spreadsheet at https://github.com/mithro/symbiflow-arch-defs/blob/v2x-latches2/utils/flipflop-generate.py

nobodywasishere, 2021-08-30:
[image]
Work in progress... Thoughts? Still need to work on the flip flops / latches. Thank you for all the information mithro!

nobodywasishere, 2021-09-24:
So I've worked a little bit more on this, specifically on the latches / flip-flops. For latches/ffs with a reset, I've just added a port onto the bottom that can be populated if it exists, and I've not distinguished between synchronous and asynchronous (yet). I also added `-bus` variants for these.

[image]

Here's an example of one diagram before:

[image]

And after:

[image]

mithro, 2021-09-25:
This looks awesome!

rishubn, 2021-11-16:
Are there plans to unroll or devise skins for complex gates within the cell library such as _AOI3_

https://github.com/YosysHQ/yosys/blob/fdb19a5b3a10572b1b84a26306ec5e83a4eff613/manual/CHAPTER_CellLib.tex#L637

nobodywasishere, 2021-11-16:
I had not planned on doing those, though I can look into it. Do you have an idea / sketch of what you think it could look like?

rishubn, 2021-11-17:
@nobodywasishere 

It would make sense to unroll these, currently their are represented as black boxes, but maybe this needs to be done on the algorithm side, not the skins

[image]

mithro, 2021-11-18:
Some suggestions from a quick google search;
 - http://www.vlsi-expert.com/2018/01/aoi-and-or-inverter-cell.html
 - https://en.wikipedia.org/wiki/AND-OR-Invert

[image]

[image]
> 2-2 AOI Symbol

mithro, 2021-11-18:
[image]

nobodywasishere, 2021-12-30:
Progress update:
- Added `$_{AND,OR}NOT_`
- Added `$_{AOI,OAI}{3,4}_`
- Added the fix in #115
- Attempted a memory cell, but not sure how good a representation this is (it's still a WIP)

[image]

mithro, 2022-01-04:
All looks pretty awesome to me!

PR #118 (2022-02-06): Add more skins
- Removes image sizes in README
- Adds cell diagrams to default.svg:
  - andnot
  - ornot
  - buf
  - pos
  - neg
  - mul
  - div
  - mod
  - pow
  - shr
  - shl
  - sshr
  - sshl
  - dlatch
  - dlatchn
  - \_AOI3\_
  - \_OAI3\_
  - \_AOI4\_
  - \_OAI4\_
- Fixes #104 
- Fixes #115 

# Comparison

Current:
[image]

New:
[image]
closed 2022-02-06 by nturley