nturley/netlistsvg #105: Wires laid out atop one another

isovector, 2021-07-03:
First of all, I just wanted to thank you all for `netlistsvg`. This is a truly fantastic piece of software!

In some cases, `netlistsvg` will run wires atop one another, eg, highlighted in red:

[image]

with source JSON:

```json
{"modules":{"module":{"ports":{},"cells":{"51":{"type":"andV","parameters":{"i0 : Bool_WIDTH":1,"i1 : Vec 4 Bool_WIDTH":4,"Vec 4 Bool_WIDTH":4},"attributes":{"name":"andV"},"port_directions":{"i0 : Bool":"input","i1 : Vec 4 Bool":"input","Vec 4 Bool":"output"},"connections":{"Vec 4 Bool":[47,48,49,50]}},"90":{"type":"andV","parameters":{"i0 : Bool_WIDTH":1,"i1 : Vec 4 Bool_WIDTH":4,"Vec 4 Bool_WIDTH":4},"attributes":{"name":"andV"},"port_directions":{"Vec 4 Bool":"output"},"connections":{"i0 : Bool":[63],"i1 : Vec 4 Bool":[73,75,77,79],"Vec 4 Bool":[86,87,88,89]}},"92":{"type":"$or","parameters":{"A_WIDTH":2},"attributes":{},"port_directions":{"A":"input","Y":"output"},"connections":{"A":[47,86],"Y":[91]}},"94":{"type":"$or","parameters":{"A_WIDTH":2},"attributes":{},"port_directions":{"A":"input","Y":"output"},"connections":{"A":[48,87],"Y":[93]}},"96":{"type":"$or","parameters":{"A_WIDTH":2},"attributes":{},"port_directions":{"A":"input","Y":"output"},"connections":{"A":[49,88],"Y":[95]}},"98":{"type":"$or","parameters":{"A_WIDTH":2},"attributes":{},"port_directions":{"A":"input","Y":"output"},"connections":{"A":[50,89],"Y":[97]}}}}}}
```

In this diagram, wires 48 and 86 are overlaid.

The above is a minimal example I pulled apart by hand. In case I butchered some invariant, the original was this:

```haskell
{"modules":{"module":{"ports":{"in0 : AluOpCode":{"direction":"input","bits":[0,1,2]},"in1 : Word4":{"direction":"input","bits":[3,4,5,6]},"in2 : Word4":{"direction":"input","bits":[7,8,9,10]},"out0 : Vec 4 Bool":{"direction":"output","bits":[91,93,95,97]}},"cells":{"35":{"type":"case AluOpAdd","parameters":{"i0 : AluOpCode_WIDTH":3,"i1 : Word4_WIDTH":4,"i2 : Word4_WIDTH":4,"o0 : Bool_WIDTH":1,"o1 : Word4_WIDTH":4,"o2 : Word4_WIDTH":4},"attributes":{"name":"case AluOpAdd"},"port_directions":{"i0 : AluOpCode":"input","i1 : Word4":"input","i2 : Word4":"input","o0 : Bool":"output","o1 : Word4":"output","o2 : Word4":"output"},"connections":{"i0 : AluOpCode":[0,1,2],"i1 : Word4":[3,4,5,6],"i2 : Word4":[7,8,9,10],"o0 : Bool":[26],"o1 : Word4":[27,28,29,30],"o2 : Word4":[31,32,33,34]}},"41":{"type":"$add","parameters":{"A_WIDTH":4,"B_WIDTH":4,"Y_WIDTH":5},"attributes":{},"port_directions":{"A":"input","B":"input","Y":"output"},"connections":{"A":[27,28,29,30],"B":[31,32,33,34],"Y":[36,37,38,39,40]}},"51":{"type":"andV","parameters":{"i0 : Bool_WIDTH":1,"i1 : Vec 4 Bool_WIDTH":4,"o0 : Vec 4 Bool_WIDTH":4},"attributes":{"name":"andV"},"port_directions":{"i0 : Bool":"input","i1 : Vec 4 Bool":"input","o0 : Vec 4 Bool":"output"},"connections":{"i0 : Bool":[26],"i1 : Vec 4 Bool":[36,37,38,39],"o0 : Vec 4 Bool":[47,48,49,50]}},"72":{"type":"case AluOpAnd","parameters":{"i0 : AluOpCode_WIDTH":3,"i1 : Word4_WIDTH":4,"i2 : Word4_WIDTH":4,"o0 : Bool_WIDTH":1,"o1 : Word4_WIDTH":4,"o2 : Word4_WIDTH":4},"attributes":{"name":"case AluOpAnd"},"port_directions":{"i0 : AluOpCode":"input","i1 : Word4":"input","i2 : Word4":"input","o0 : Bool":"output","o1 : Word4":"output","o2 : Word4":"output"},"connections":{"i0 : AluOpCode":[0,1,2],"i1 : Word4":[3,4,5,6],"i2 : Word4":[7,8,9,10],"o0 : Bool":[63],"o1 : Word4":[64,65,66,67],"o2 : Word4":[68,69,70,71]}},"74":{"type":"$and","parameters":{"A_WIDTH":1,"B_WIDTH":1,"Y_WIDTH":1},"attributes":{},"port_directions":{"A":"input","B":"input","Y":"output"},"connections":{"A":[64],"B":[68],"Y":[73]}},"76":{"type":"$and","parameters":{"A_WIDTH":1,"B_WIDTH":1,"Y_WIDTH":1},"attributes":{},"port_directions":{"A":"input","B":"input","Y":"output"},"connections":{"A":[65],"B":[69],"Y":[75]}},"78":{"type":"$and","parameters":{"A_WIDTH":1,"B_WIDTH":1,"Y_WIDTH":1},"attributes":{},"port_directions":{"A":"input","B":"input","Y":"output"},"connections":{"A":[66],"B":[70],"Y":[77]}},"80":{"type":"$and","parameters":{"A_WIDTH":1,"B_WIDTH":1,"Y_WIDTH":1},"attributes":{},"port_directions":{"A":"input","B":"input","Y":"output"},"connections":{"A":[67],"B":[71],"Y":[79]}},"90":{"type":"andV","parameters":{"i0 : Bool_WIDTH":1,"i1 : Vec 4 Bool_WIDTH":4,"o0 : Vec 4 Bool_WIDTH":4},"attributes":{"name":"andV"},"port_directions":{"i0 : Bool":"input","i1 : Vec 4 Bool":"input","o0 : Vec 4 Bool":"output"},"connections":{"i0 : Bool":[63],"i1 : Vec 4 Bool":[73,75,77,79],"o0 : Vec 4 Bool":[86,87,88,89]}},"92":{"type":"$or","parameters":{"A_WIDTH":2},"attributes":{},"port_directions":{"A":"input","Y":"output"},"connections":{"A":[47,86],"Y":[91]}},"94":{"type":"$or","parameters":{"A_WIDTH":2},"attributes":{},"port_directions":{"A":"input","Y":"output"},"connections":{"A":[48,87],"Y":[93]}},"96":{"type":"$or","parameters":{"A_WIDTH":2},"attributes":{},"port_directions":{"A":"input","Y":"output"},"connections":{"A":[49,88],"Y":[95]}},"98":{"type":"$or","parameters":{"A_WIDTH":2},"attributes":{},"port_directions":{"A":"input","Y":"output"},"connections":{"A":[50,89],"Y":[97]}}}}}}
```

which also exhibits the bug.

nobodywasishere, 2021-07-03:
This was an underlying problem within ELK itself [here](https://github.com/kieler/elkjs/issues/77), but has since been fixed. Updating the version of ELKjs used by netlistsvg from 0.3.0 to 0.7.1 (the latest) didn't seem to break anything and it also fixed the issue you're experiencing. 

[image]

isovector, 2021-07-03:
Thanks so much!

closed 2021-07-05 by nturley