nturley/netlistsvg #111: Can not draw svg with a "hardcoded FPGA block"

71GA, 2021-09-22:
I have this design where I use SPI block that is "hardcoded inside the FPGA". I don't have the design of this element, but I am able to use it successfully once my bitstream file is uploaded to the FPGA.

This is my design _(in Verilog)_:

```
// A:
module
	top(
		input pcb_oscilator,
		output pcb_rgb_led0,
		output pcb_rgb_led1,
		output pcb_rgb_led2,
		output pcb_led1,
		output pcb_led2,
		output pcb_led3,
		output pcb_led4,
		output pcb_led5,
		input pcb_spi_sck,
		input pcb_spi_csn,
		input pcb_spi_si,
		output pcb_spi_so
);

	// B:
	parameter
		NOP = 0,
		INIT = 1,
		WR_INVERTED = 2,
		WR_LEDS = 4,
		WR_VEC = 6,
		RD_VEC = 7;

	// C:
	parameter
		INIT_SPICR0 = 0,
		INIT_SPICR1 = 1,
		INIT_SPICR2 = 2,
		INIT_SPIBR = 3,
		INIT_SPICSR = 4,
		SPI_WAIT_RECEPTION = 5,
		SPI_READ_OPCODE = 6,
		SPI_READ_LED_VALUE = 7,
		SPI_READ_INIT = 8,
		SPI_SEND_DATA = 9,
		SPI_WAIT_TRANSMIT_READY = 10,
		SPI_TRANSMIT = 11,
		STATEA = 12,
		LOOPSTATE = 13;

	// D:
	parameter
		SPI_ADDR_SPICR0 = 8'b00001000,
		SPI_ADDR_SPICR1 = 8'b00001001,
		SPI_ADDR_SPICR2 = 8'b00001010,
		SPI_ADDR_SPIBR = 8'b00001011,
		SPI_ADDR_SPITXDR = 8'b00001101,
		SPI_ADDR_SPIRXDR = 8'b00001110,
		SPI_ADDR_SPICSR = 8'b00001111,
		SPI_ADDR_SPISR = 8'b00001100;
		/// SPI_ADDR_SPIINTSR = 8'b00000110,
		/// SPI_ADDR_SPIINTCR = 8'b00000111;

	// E:
	///parameter COMMAND_SIZE = 4;

	// F:
	wire sb_clki;
	reg sb_rwi;
	reg sb_stbi;
	reg [7:0] sb_adri;
[... 260 lines cut ...]
        automatically determine the top of the design hierarchy and mark it.

    -chparam name value 
       elaborate the top module using this parameter value. Modules on which
       this parameter does not exist may cause a warning message to be output.
       This option can be specified multiple times to override multiple
       parameters. String values must be passed in double quotes (").

In -generate mode this pass generates blackbox modules for the given cell
types (wildcards supported). For this the design is searched for cells that
match the given types and then the given port declarations are used to
determine the direction of the ports. The syntax for a port declaration is:

    {i|o|io}[@<num>]:<portname>

Input ports are specified with the 'i' prefix, output ports with the 'o'
prefix and inout ports with the 'io' prefix. The optional <num> specifies
the position of the port in the parameter list (needed when instantiated
using positional arguments). When <num> is not specified, the <portname> can
also contain wildcard characters.

This pass ignores the current selection and always operates on all modules
in the current design.
```
This problem is related to `yosys` and `netlistsvg`, so I posted it here.

71GA, 2021-09-22:
I got solution here: https://github.com/YosysHQ/yosys/issues/3013

nobodywasishere, 2021-09-24:
This is a problem strictly with yosys and not netlistsvg.

closed 2021-09-27 by nturley