nturley/netlistsvg #112: Exported svg is using "physical wiring connections" instead of "signal name association" (netlists)

71GA, 2021-09-22:
I sucessfully converted my design in a `.svg` file but I see that all the connections are drawn using the _"physicall wiring connections"_ like this:
[image]

And this is extremely hard to read unlike _"netlists"_: 
[image]

Is there a way to add support for _"netlists"_? So that schematics will at least be useful in development.

nobodywasishere, 2021-09-24:
I'm not sure if there'd be an easy way to determine which wires to disconnect and replace with labels vs which wires to leave in place. One option could just be opening up the generated netlist in inkscape afterwards and changing it around to fit what you're looking for, but this would be more of a one-time thing. 

Personally, I think the first image you linked is easier to read (and what I would refer to as a netlist), but that's just me. I don't have to make mental connections of how everything fits together.

rjordans, 2021-09-25:
I could imagine having a list of nets that should be translated into labels wherever they are used.  That might be useful for more 'global' nets like clock or reset lines that can go in all kinds of directions sometimes.  For most of the other nets I'd prefer the first image version too.

71GA, 2021-10-05:
@nobodywasishere 1st image is "easier to" read exclusively because it is a really small design. Try this still "small" design of mine.

[top.pdf](https://github.com/nturley/netlistsvg/files/7284746/top.pdf)

Good luck...

Probably now you see... It is impossible to read it, because vertical wires clutter everything. Try to follow some wires... You might even get dizzy!

71GA, 2021-10-05:
Also in large designs it is hard to follow a wire, but when you have a label, you can simply <kbd>CTRL</kbd>+<kbd>F</kbd> and be done. Maybee it is possible to at least somehow implement an algorythm that checks the length of the wires and replaces the long ones with the labels. The long ones are the ones that mess everything anyway...