nturley/netlistsvg #126: How to decode this error message?

saicharan0112, 2022-09-10:
I am trying to generate the svg file from a yosys exported json file using the netlistsvg version that is available on the npm. I get this error but I don't understand anything. Can anyone help me understand what it says from the netlistsvg perspective please?

Here is the msg. 
[netlistsvg_error.log](https://github.com/nturley/netlistsvg/files/9540862/netlistsvg_error.log)

TheZoq2, 2022-09-30:
I experienced this as well. Installing the latest version from git using `npm install -g git@github.com:nturley/netlistsvg.git` as described in #120 fixed it for me.

gpelisset, 2022-12-17:
I'm still experiencing this problem when trying to draw a simple buffer tristate. Here is the error
/home/gp/node_modules/netlistsvg/bin/netlistsvg.js:55
            throw Error(JSON.stringify(ajv.errors, null, 2));
            ^

Error: [
  {
    "keyword": "type",
    "dataPath": "/modules/buffer_tristate/cells/$auto$simplemap.cc:278:simplemap_mux$79/connections/A/0",
    "schemaPath": "#/properties/modules/additionalProperties/properties/cells/additionalProperties/properties/connections/additionalProperties/items/oneOf/0/type",
    "params": {
      "type": "number"
    },
    "message": "should be number"
  },
  {
    "keyword": "enum",
    "dataPath": "/modules/buffer_tristate/cells/$auto$simplemap.cc:278:simplemap_mux$79/connections/A/0",
    "schemaPath": "#/properties/modules/additionalProperties/properties/cells/additionalProperties/properties/connections/additionalProperties/items/oneOf/1/enum",
    "params": {
      "allowedValues": [
        "0",
        "1",
        "x"
      ]
    },
    "message": "should be equal to one of the allowed values"
  },
  {
    "keyword": "oneOf",
    "dataPath": "/modules/buffer_tristate/cells/$auto$simplemap.cc:278:simplemap_mux$79/connections/A/0",
    "schemaPath": "#/properties/modules/additionalProperties/properties/cells/additionalProperties/properties/connections/additionalProperties/items/oneOf",
    "params": {
      "passingSchemas": null
    },
    "message": "should match exactly one schema in oneOf"
  }
]
    at /home/gp/node_modules/netlistsvg/bin/netlistsvg.js:55:19
    at /home/gp/node_modules/netlistsvg/bin/netlistsvg.js:41:17
    at FSReqCallback.readFileAfterClose [as oncomplete] (node:internal/fs/read_file_context:68:3)

Node.js v19.2.0.

This is the verilog file:
module buffer_tristate(input  logic a, en,
                       output tri   y);
    assign y = en ? a : 1'bz;
endmodule

Any hope to see it fixed?