nturley/netlistsvg #128: $_ANDNOT gate is rendered as $_NAND

alexandre-frey, 2023-01-04:
The attached JSON defines a single $_ANDNOT gate. But it gets rendered as a NAND gate instead. See attached SVG.
netlistsvg version is: 1.0.2
[netlistsvg_andnot_vs_nand.zip](https://github.com/nturley/netlistsvg/files/10346244/netlistsvg_andnot_vs_nand.zip)

Popolon, 2023-02-19:
What is the difference between NAND and AND NOT? Do you have some link about it, or an example of correct representation? NAND is already a and with the NOT a the output, this is not really clear.

nobodywasishere, 2023-02-19:
@Popolon `andnot` is an and gate with one of the inputs inverted, see the current skin file:

[image]

Popolon, 2023-02-23:
Thank you very much :) I was unable to found the answer. And nobody known around me, hope it will be a good reference on this point.