nturley/netlistsvg #140: Feature Request: Generate arithmetic operations fused with constants

sifferman, 2024-01-04:
Hello.

DigitalJS has a really cool feature to improve schematic readability by generating arithmetic operations fused with constants: [code source](https://github.com/tilk/digitaljs/blob/a541f0ff32b4560e593cfb6cf5b5eed06b902fec/src/cells/arith.mjs#L414-L645). I believe this would be a very useful addition to netlistsvg as well.

For reference, here are schematics generated by netlistsvg and [DigitalJS Online](https://digitaljs.tilk.eu/) of a basic register file. DigitalJS' schematic is much more readable, likely due to the addition of arithmetic operations fused with constants.

[image]
[image]

mithro, 2024-01-04:
@sifferman - Random side question -- do you think the DigitalJS Online stuff could be integrated into [sphinxcontrib-hdl-diagrams](https://github.com/SymbiFlow/sphinxcontrib-hdl-diagrams/) ? It currently supports both Yosys diagrams and NetlistSVG diagrams.

pidgeon777, 2024-02-22:
I'm also interested